unihd-cag/skillbridge #251: [SUPPORT] How to Execute SKILL Code Using evalstring Function with Skillbridge

Galac-hpw, 2024-03-18:
I'm currently working on integrating Python scripts with Cadence Virtuoso using the Skillbridge library. My goal is to enhance my automation capabilities within Virtuoso by leveraging Python's versatility. Specifically, I am interested in executing arbitrary SKILL code strings from Python, which typically involves the evalstring function in the SKILL language environment.

However, after going through the Skillbridge documentation and various examples ([Skillbridge Documentation](https://unihd-cag.github.io/skillbridge/)), I couldn't find explicit guidance on how to directly call the evalstring function or a similar method that allows for executing raw SKILL code strings through the Skillbridge interface. Or simply allow a piece of skill code in textual form.

I'm reaching out to see if anyone has successfully used Skillbridge for this purpose and can share insights or examples on how to implement this. Is there a direct method or workaround within Skillbridge that facilitates the execution of SKILL code strings through Python? Any guidance or suggestions on this matter would be greatly appreciated, as it would significantly enhance the automation processes I am developing for Virtuoso.

Thank you in advance for your help and time!

nielsbuwen, 2024-03-18:
You can call arbitrary SKILL function like this:

```python
from skillbridge import Workspace

ws = Workspace.open()

result = ws['evalstring']("1 + 2")
print(result)
```

Galac-hpw, 2024-03-19:
Thank you so much for your help in resolving this issue!

closed 2024-03-19 by Galac-hpw