unihd-cag/skillbridge #252: How to use underlying api

xiaohangguo, 2024-03-29:
I'd like to ask how to call the underlying api like` SymbolToString` in python.

nielsbuwen, 2024-03-29:
In Python a `Symbol` is just a named tuple and you can extract the string value like this:

```python
my_symbol = some_function_that_returns_symbol(...)

print(my_symbol.name)
```

To call arbitrary functions you can also use the `Workspace` like this:

```python
symbol_name = ws['symbolToString'](my_symbol)
```

xiaohangguo, 2024-03-29:
Thanks！！！！I didn't know you could just use that. @nielsbuwen 
`ws['xxxx'](arg)`

Thank you very much, I found a feature where I can write custom il files in skill, implement special functions and then access them that way.
wow!

I'm working on rewriting il files to python interfaces and this feature will help me reduce the amount of code, great!

closed 2024-03-29 by xiaohangguo