unihd-cag/skillbridge #256: [SUPPORT] Multiple communications from one device

xiaohangguo, 2024-04-07:
How do I get multiple programs to use this communication skillgridge on a single machine?
I tried it and found that it would have sockets, changing the name didn't help either

TM90, 2024-04-07:
Hello,

Having multiple programs accessing the server is possible as long as they are working sequentially. You should use `ws = Workspace.open()` and `ws.close()` accordingly.

If you want to have multiple applications running that access virtuoso at any given time, this will be a problem since a lot of virtuosos commands will change state and you will get inconsistencies.

Another approach would be that each application spawns its virtuoso instance with a unique id.

I hope this helps.

Best regards

Tobias

xiaohangguo, 2024-04-08:
> Hello,
>
> Having multiple programs accessing the server is possible as long as they are working sequentially. You should use `ws = Workspace.open()` and `ws.close()` accordingly.
>
> If you want to have [...]

hello！ thanks . 
i use the command ，But it didn't work.
`pyStartServer ?id "lvshuhang"`
 

On the server I am using 'pyStartServer' which is working fine, now I want to set up different communication services on the same device so that I can have multiple users for running and communicating with their respective python-virtuoso programs. But now after I start it with the command `pyStartServer ?id "lvshuhang"`, it doesn't work at all!

TM90, 2024-04-08:
Can you provide a simplified/minimal example of your use case?

TM90, 2024-04-08:
In your second example how does your client open the workspace ?

Maybe this example is helpful:

https://unihd-cag.github.io/skillbridge/examples/multiple-instances.html

xiaohangguo, 2024-04-08:
i got it,That's how I should
`ws = Workspace.open('lvshuhang')`

xiaohangguo, 2024-04-08:
[image]
I'm trying to do it in two python terminals and in a virtuoso terminal, and it's still failing.

xiaohangguo, 2024-04-08:
I'm testing on the same server now, I have two accounts 'lvshuhang' and 'wuqingsen_s07'

```
(py390) 
lvshuhang@s07:/tmp
$ ls -l skill-server-lvshuhang.sock
srwxrwxr-x 1 lvshuhang lvshuhang 0 Apr  8 16:59 skill-server-lvshuhang.sock
(py390) 
lvshuhang@s07:/tmp
$ ls -l skill-server-wuqingsen_s07-wqs.sock
srwxrwxr-x 1 wuqingsen_s07 wuqingsen_s07 0 Apr  8 17:32 skill-server-wuqingsen_s07-wqs.sock
```

```
wuqingsen_s07@s07:~
$ ls -l /tmp/skill-server-wuqingsen_s07-wqs.sock
srwxrwxr-x 1 wuqingsen_s07 wuqingsen_s07 0 Apr  8 17:32 /tmp/skill-server-wuqingsen_s07-wqs.sock
(py390) 
wuqingsen_s07@s07:~
$ 
```

Both users should have permission to read and write their own sock, but why is it that user 'wuqingsen_s07' can't run properly and reports a permission error?

output:
```
$ s180bcd ./ted-viewer/core/platform/virtuoso/awv_plot/tests/test_1.py
=========================================================== test session starts ===========================================================
platform linux -- Python 3.9.0, pytest-7.2.0, pluggy-1.0.0 -- /EDA/soft/anaconda3.7/envs/py390/bin/python
cachedir: .pytest_cache
rootdir: /home/wuqingsen_s07
plugins: typeguard-2.13.3, jaxtyping-0.2.28, regressions-2.4.1, xdist-3.1.0, anyio-3.6.2, cov-4.0.0, datadir-1.4.1, timeout-2.1.0
collected 1 item                                                                                                                          

ted-viewer/core/platform/virtuoso/awv_plot/tests/test_1.py::test_awvPlot FAILED

================================================================ FAILURES =================================================================
______________________________________________________________ test_awvPlot _______________________________________________________________

    def test_awvPlot():
>       awv_plot = VirtuosoAwvPlot()

ted-viewer/core/platform/virtuoso/awv_plot/tests/test_1.py:6: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
ted-viewer/core/platform/virtuoso/awv_plot/plot.py:25: in __init__
    super().__init__(work_space, cell_view)
ted-viewer/core/platform/virtuoso/base/virtuoso_base.py:21: in __init__
    self.work_space = Workspace.open()
.local/lib/python3.9/site-packages/skillbridge/client/workspace.py:268: in open
    channel = channel_class(workspace_id)
.local/lib/python3.9/site-packages/skillbridge/client/channel.py:80: in __init__
    self.socket = self.start()
.local/lib/python3.9/site-packages/skillbridge/client/channel.py:89: in start
    return self.connect(sock)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <skillbridge.client.channel.create_channel_class.<locals>.UnixChannel object at 0x7ff11cb104f0>
sock = <socket.socket fd=7, family=AddressFamily.AF_UNIX, type=SocketKind.SOCK_STREAM, proto=0>

    def connect(self, sock: socket) -> socket:
        sock.settimeout(1)
>       sock.connect(self.address)
E       PermissionError: [Errno 13] Permission denied

.local/lib/python3.9/site-packages/skillbridge/client/channel.py:99: PermissionError
========================================================= short test summary info =========================================================
FAILED ted-viewer/core/platform/virtuoso/awv_plot/tests/test_1.py::test_awvPlot - PermissionError: [Errno 13] Permission denied
============================================================ 1 failed in 0.28s =====================================================
```

nielsbuwen, 2024-04-08:
Did each user start their own Virtuoso session?

TM90, 2024-04-08:
In your 

> > Did each user start their own Virtuoso session?
> > The permissions look like access is only allowed to the same group? Do the socket files belong to the shared group?
> 
> Each user starts their own virtuoso, and socket permissions should only run their own user access to its socket
> 
> ![image](https://private-user-images.githubusercontent.com/72799392/320515309-4d45cd8c-baee-4bc8-8eff-82146cf6f578.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1ODYwODksIm5iZiI6MTcxMjU4NTc4OSwicGF0aCI6Ii83Mjc5OTM5Mi8zMjA1MTUzMDktNGQ0NWNkOGMtYmFlZS00YmM4LThlZmYtODIxNDZjZjZmNTc4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA4VDE0MTYyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA5ZDc2ZGU1NjJmMzc5ZWI4ZmY4ODY5YzhlMWZlZjBiZjNlNDlkNzM5NTc3OWZlNjNjYzhmNDAzOTk5ZjU3Y2EmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Gy1k-9aKMNieviGkI-B3Zsuwgw2CryT_ffizjfITfZA)
> 
> ![image](https://private-user-images.githubusercontent.com/72799392/320515127-47936c0f-b0be-4b48-857a-a25535f16186.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI1ODYwODksIm5iZiI6MTcxMjU4NTc4OSwicGF0aCI6Ii83Mjc5OTM5Mi8zMjA1MTUxMjctNDc5MzZjMGYtYjBiZS00YjQ4LTg1N2EtYTI1NTM1ZjE2MTg2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA4VDE0MTYyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMxMWE0OWRlZTY0YWVmZTNlY2RiODlkODc1MjllZjA0NWJhNzM0ZWQ5ZWM0ZTQ5OTM3MTNmZjVjNTRhNWRmZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.5z6uISzyEhALCJl8Y-ZUAAnPiTuUsMjy_TIG1HrzwVA)

In the shown trace you just call `Workspace.open()` without the correct id.

xiaohangguo, 2024-04-08:
> In your
>
>
> In the shown trace you just call `Workspace.open()` without the correct id.

no,not this problem. I've tried them all.

[image]

TM90, 2024-04-08:
Can you repeat this in a clean session. Maybe the testing lead to some leftover processes?

nielsbuwen, 2024-04-08:
Just to make sure we are on the same page. You did the following, right?

- UserA on machine X starts Virtuoso
- UserA runs `pyStartServer ?id userA`
- UserA runs `Workspace.open('userA')`
- UserB on machine X starts Virtuoso
- UserB runs `pyStartServer ?id userB`
- UserB runs `Workspace.open('userB')`

We are using that exact setup and it works, correct @TM90 ?

xiaohangguo, 2024-04-08:
> Can you repeat this in a clean session. Maybe the testing led to some leftover processes?

how to clean? 
```
I tried killing the virtuoso terminal, reopening it, and running the
load("/home/wuqingsen_s07/.local/lib/python3.9/site-packages/skillbridge/server/python_server.il")
pyStartServer ?id "wqs"
```

but still :
```
================================================================ FAILURES =================================================================
______________________________________________________________ test_awvPlot _______________________________________________________________

    def test_awvPlot():
        #awv_plot = VirtuosoAwvPlot()
>       ws = Workspace.open("wqs")  # 可以理解为 打开的某个virtuoso终端

ted-viewer/core/platform/virtuoso/awv_plot/tests/test_1.py:7: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
.local/lib/python3.9/site-packages/skillbridge/client/workspace.py:268: in open
    channel = channel_class(workspace_id)
.local/lib/python3.9/site-packages/skillbridge/client/channel.py:80: in __init__
    self.socket = self.start()
.local/lib/python3.9/site-packages/skillbridge/client/channel.py:89: in start
    return self.connect(sock)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <skillbridge.client.channel.create_channel_class.<locals>.UnixChannel object at 0x7f0377aef4c0>
sock = <socket.socket fd=7, family=AddressFamily.AF_UNIX, type=SocketKind.SOCK_STREAM, proto=0>

    def connect(self, sock: socket) -> socket:
        sock.settimeout(1)
>       sock.connect(self.address)
E       PermissionError: [Errno 13] Permission denied

.local/lib/python3.9/site-packages/skillbridge/client/channel.py:99: PermissionError
========================================================= short test summary info =========================================================
FAILED ted-viewer/core/platform/virtuoso/awv_plot/tests/test_1.py::test_awvPlot - PermissionError: [Errno 13] Permission denied
============================================================ 1 failed in 0.24s ============================================================
```

nielsbuwen, 2024-04-08:
Maybe some leftover socket files cause a problem, you can safely delete them

xiaohangguo, 2024-04-08:
> Just to make sure we are on the same page. You did the following, right?
>
> * UserA on machine X starts Virtuoso
> * UserA runs `pyStartServer ?id userA`
> * UserA runs `Workspace.open('userA')`
> * UserB [...]

yes,Now my resultant surface, userA can be allowed normally, userB will have PermissionError: [Errno 13] Permission denied

xiaohangguo, 2024-04-08:
i delete all the `skill*` file ,and ,run again.
but:
[image]

[image]

nielsbuwen, 2024-04-08:
Somethign is weird on your side. Running `pyStartServer ?id "wqs"` should create `skill-server-wqs.sock` but your file is calles `skill-server-wuqingsen_s07-wqs.sock`. That is why `Workspace.open("wqs")` fails with that error.

Did you rename the `.sock` file? It shouldn't be named like that

xiaohangguo, 2024-04-08:
> Something is weird on your side. Running `pyStartServer ?id "wqs"` should create `skill-server-wqs.sock` but your file is called `skill-server-wuqingsen_s07-wqs.sock`. That is why `Workspace.open("wqs")` fails with that error.
> 
> Did you rename the `.sock` file? It shouldn't be named like that

> Something is weird on your side. Running `pyStartServer ?id "wqs"` should create `skill-server-wqs.sock` but your file is called `skill-server-wuqingsen_s07-wqs.sock`. That is why `Workspace.open("wqs")` fails with that error.
> 
> Did you rename the `.sock` file? It shouldn't be named like that

ok! nice, solve it.
It seems that I had modified the source sock naming to add os.get_login(), and looking at the documentation later, I found that it had indeed been implemented. I re-installed and reinstalled skillbriadge just fine.

thanks @TM90  @nielsbuwen 
You're all enthusiastic.

xiaohangguo, 2024-04-08:
I changed the source code on userA, reinstalled it later, and changed it on userB, but userB forgot to reinstall it, so I never noticed the problem

closed 2024-04-08 by xiaohangguo