unihd-cag/skillbridge #257: Why did I encounter the "RuntimeError: The server unexpectedly died" error?

zephyruswqs, 2024-04-11:
I am testing using outputting() to receive strings output by virtuoso. According to the example in virtuoso's function documentation, I need to use close() to close the port of oustring() after using getOutstring().

The problem now is that after testing in Python, it was found that using close() to close oustring() would result in a "RuntimeError: The server unexpectedly died" error. May I ask if there is a problem with my operation?

* Example of document:
[image]

*My code:
[image]

*Error message:
[image]

nielsbuwen, 2024-04-11:
This seems to be indeed a bug on our side. I'm investigating the issue

nielsbuwen, 2024-04-11:
The command for the `close(...)` function call was mistaken for an internal `close` command which shuts down the connection of the Workspace with the server. The linked PR fixes this.

When it is merged, we will deploy a release

TM90, 2024-04-11:
Fixed in skillbridge [1.7.1 release](https://github.com/unihd-cag/skillbridge/releases/tag/releases%2F1.7.1)

To update use `pip install --upgrade skillbridge`

closed 2024-04-11 by TM90

nielsbuwen, 2024-04-11:
The release has been deployed as version `1.7.1`

zephyruswqs, 2024-04-11:
> The command for the `close(...)` function call was mistaken for an internal `close` command which shuts down the connection of the Workspace with the server. The linked PR fixes this.
> 
> When it is merged, we will deploy a release

Okay, thank you. It's working normally now.

PR #258 (2024-04-11): Fix internal close command collision
Internally we use the string `close` to close the connection of a workspace with the server. Unfortunately this is also a valid python request.

This PR changes the internal command to `$close`. This is neither valid Python code nor valid SKILL code so it cannot be sent by accident.

Closes #257