unihd-cag/skillbridge #259: Extend skillbridge to multi-user, multi-node environments

omythic, 2024-04-18:
Skillbridge is awesome - thank you!

This issue centers around conflicts over the skillbridge Unix socket file in /tmp -- the default will conflict among users on the same node, and /tmp won't be the same on different nodes, in general -- consider Virtuoso being launched to a compute node. Could it optionally be in a /home dot-dir? (XDG config?)

Avoiding a file altogether, could a TCP connection (host, port) be safely used in some way (somehow exposed/confined to single user)?

nielsbuwen, 2024-04-18:
We decided against a TCP connection, because Unix socket was faster in our measurements and it prevents other users from interfering with your server.

We can, of course, put the socket file somewhere else. We just have to check whether that breaks anything.

omythic, 2024-04-18:
Thanks for your speedy reply!

So, putting the socket file elsewhere (like $HOME) will definitely solve the user-contention-same-node aspect.

But multi-host is still not an option, if I understand this correctly: https://superuser.com/questions/1608873/does-a-unix-domain-socket-in-an-nfs-exported-fs-magically-work-across-hosts

So a user will not be able to connect from his local node to a Virtuoso session launched to another node. Do I understand that right?

I'd really like to play with a TCP connection -- you don't happen to have a branch of skillbridge where you experimented with host/port connection?

Much appreciated!

TM90, 2024-04-19:
> Thanks for your speedy reply!
>
> So, putting the socket file elsewhere (like $HOME) will definitely solve the user-contention-same-node aspect.
>
> But multi-host is still not an option, if I understand [...]

Take a look in https://github.com/unihd-cag/skillbridge/tree/master/skillbridge/server

The Windows variant of the server actually runs TCP on localhost. Maybe this is a good starting point.

Regarding the socket file if there is a change for the location it should be optionally configurable.

omythic, 2024-04-19:
Cool, I had seen that but was being lazy ;)

I guess you could if you like convert this issue into "configurable unix socket file". Thanks again!

closed 2024-04-19 by omythic