unihd-cag/skillbridge #261: Extract x and y vectors from an srrWave [SUPPORT]

mvosta, 2024-04-30:
The goal is to extract the x and y vectors from an srrWave. In a first step from a single design point, but in a later stage ideally from a parameter sweep simulation. The current approach to get the x and y vectors from a waveform from a node voltage of a transient simulation is shown below.

The function to convert a waveform to two vectors:
```
def wave2vec(ws, waveform):
    waveform = ws.dr.get_waveform_y_vec(waveform)

    x_vec = ws.dr.get_waveform_x_vec(waveform)
    x = []
    for i in range(ws.dr.vector_length(x_vec)):
        x.append(ws.dr.get_elem(x_vec, i))
    
    y_vec = ws.dr.get_waveform_y_vec(waveform)
    y = []
    for i in range(ws.dr.vector_length(y_vec)):
        y.append(ws.dr.get_elem(y_vec, i))
    return x, y
```
Part of the code where the function is used:
```
# setup Maestro mode explorer
ws['ocnSetXLMode']("explorer") # set mode to OCEANXL
ws['ocnxlProjectDir']("~/simulations") # location of simulation results
ws['ocnxlTargetCellView']("library", "cell", "view")
ws['ocnxlResultsLocation']("") # same as project dir
ws['ocnxlSimResultsLocation']("") # same as project dir
ws['ocnxlMaxJobFail'](20)

ws['openResults'](path_data)
ws['selectResult'](Symbol("tran"))
data = ws['getData']("/net1")

x, y = wave2vec(ws, data)

print(x)
print(y)

ws['ocnxlEndXLMode']('explorer')
```
The printed results are:
 - `[1.0]`
 - `[<remote object@0x...>]`

Which is not what I would expect. How should the approach above be altered to successfully get the x and y vectors?

nielsbuwen, 2024-04-30:
Hi, can you provide us with the working SKILL code, please? Then we can take a look why the translation does not work

mvosta, 2024-04-30:
Hi, thank you for the quick response. The problem was apparently not related to Skill Bridge, but to a wrong path for the openResults function.

mvosta, 2024-04-30:
Hi, thank you for the quick response. The issue was apparently due to a wrong path for the `openResults` function. The path was set to `"/somepath/Ocean.0/psf/tran/psf/"`, which points to the root of the results containing multiple simulation points. This path should point to `"/somepath/Ocean.0/1/tran/psf/"`, i.e. a single simulation point, which solves the issue.

closed 2024-04-30 by mvosta