unihd-cag/skillbridge #262: [SUPPORT] instantiate an instance with create_inst_by_master_name. Invalid origin error

zh1609, 2024-05-15:
I am having issue to instantiate a mos device. The skill and python code below. I am getting RuntimeError:(dbCreateInstByMasterName" 0 t nil ("*Error* dbCreateInstByMasterName: Invalid origin" ((0.0 0.0))))

Skill
``
    dbCreateInstByMasterName(cv "basic" "nand2" "symbol" "inst1" list(0 0) "R0" 1)
``

Python
``
 ws=Workspace.open()                          
 #create nand cell view in basic lib.     
i=ws.open_cell_view_by_type("basic","nand2","schematic")
cell_view=ws.db.open_cell_view("basic","nand2","schematic")                                    
#instantiate a mosfet in nand2
i=ws.db.create_inst_by_master_name(cell_view,"basic","pmos","symbol","M0",[0.0,0.0],"R0",1)
``

nielsbuwen, 2024-05-15:
You put double braces around the origin. `[[0.0, 0.0]]` creates a list with a single element which is the origin. But you want a flat list of x/y coordinates: `[0.0, 0.0]`

```python
ws.db.create_inst_by_master_name(cell_view, "basic", "pmos", "symbol", "M0", [0.0, 0.0], "R0", 1)
```

zh1609, 2024-05-15:
I tried with the single bracket, its giving another warning
Skill response:0: UserWarning:  (DB-270330): dbCreateInstByMasterName: cellview recursion detected. :(

closed 2024-05-15 by zh1609