unihd-cag/skillbridge #263: [SUPPORT] Help understanding -> and ~> in Skillbridge

mkaikkon, 2024-05-15:
Hello, I am still in the early steps of using this nice platform. I have some general knowledge of SKILL and am able to get things working in the CIW most of the time, but there are some things I have not been able to figure out on my own.

The one I am most having most confusion with is how to call the ~> operator in Skillbridge. As far as I understand, I can call the -> operator by using a dot, for example to get the prop ID of an inductor "IND":

`ws = Workspace.open()
cell_view = ws.db.open_cell_view("Library", "Inductor_Test", "schematic")
indInst = ws.db.get_instance_by_name(cell_view, "IND")
indProp = indInst.prop`

This would return me the property ID's of the instance "IND". Next what I would like to do is read the names of these instances. In SKILL, I would first of course have the SKILL equivalent of the previous code, but the follow it up with:

`indProp~>name`

And the CIW prints out the names of all the variables. However, it is very important that I use ~> instead of -> to get the result that I want, as using -> just ends up returning a nil (My assumption is that -> is for returning object ID's and ~> is for returning the actual values, or something along those lines). 

In Skillbridge I have not been able to figure out how to use the ~> operator, and as I explained -> does not return what I want. This sounds like a very basic thing so I apologize if I missed this in some of the previous questions. I could not find an answer in the documentation either.

TM90, 2024-05-16:
There is not a single operator in python doing the same as `indProp~>name`.

But you can achieve the same with a list comprehension:

Try:
```
ws = Workspace.open()
cell_view = ws.db.open_cell_view("Library", "Inductor_Test", "schematic")
ind_inst = ws.db.get_instance_by_name(cell_view, "IND")
ind_props = indInst.prop # This seems to be a list[RemoteObjects] each element containing all properties
ind_names = [prop.name for prop in ind_props] # This will create a new list containing all names
```

mkaikkon, 2024-05-16:
Hi,

Thank you! I got it! In addition to what you wrote, I needed to wrap the generator expression with a list() function or else it would return a generator object <genexpr> instead of the list of names:
```
ws = Workspace.open()
cell_view = ws.db.open_cell_view("Library", "Inductor_Test", "schematic")
ind_inst = ws.db.get_instance_by_name(cell_view, "IND")
ind_props = indInst.prop # This seems to be a list[RemoteObjects] each element containing all properties
ind_names = ( list (prop.name for prop in ind_props) ) # This will create a new list containing all names
```

And if I was trying to get numerical values like the prop values, I would wrap the generator with a tuple(). 

Thanks for the help!

closed 2024-05-16 by mkaikkon

TM90, 2024-05-16:
> Hi,
> 
> Thank you! I got it! In addition to what you wrote, I needed to wrap the generator expression with a list() function or else it would return a generator object instead of the list of names:
> 
> ```
> ws = Workspace.open()
> cell_view = ws.db.open_cell_view("Library", "Inductor_Test", "schematic")
> ind_inst = ws.db.get_instance_by_name(cell_view, "IND")
> ind_props = indInst.prop # This seems to be a list[RemoteObjects] each element containing all properties
> ind_names = ( list (prop.name for prop in ind_props) ) # This will create a new list containing all names
> ```
> 
> And if I was trying to get numerical values like the prop values, I would wrap the generator with a tuple().
> 
> Thanks for the help!

If you use a list with the square brackets as described above you will direclty get a list ;)