unihd-cag/skillbridge #265: [SUPPORT] leHiCreateInst must  interactive

xiaohangguo, 2024-06-07:
I would like to get the value of the mom capacitor by creating instances, but I can't do it this way yet, I have to do an interaction in virtuoso to make the program go on properly, is there a good way to do it?
```
    ws['evalstring']("leHiCreateInst()")
    ws['evalstring']("hiiSetCurrentForm('leCreateInstXLForm)")
    ws['evalstring']("leCreateInstXLForm->libName->value=\"hlmc55ef_hv\"")
    ws['evalstring']("leCreateInstXLForm->cellName->value=\"mom12\"")
    # ws['load']("/home/lvshuhang/ted-viewer/core/interface/awv_plot/tests/temp.il")
    c = ws['evalstring']("leCreateInstXLForm->c->value")
    debug(c)
    # ws["hiiToggleEnterForm"]("leCreateInstXLForm") 
    # ws["addPoint"]([0,0])
    ws["cancelEnterFun"]()
```

nielsbuwen, 2024-06-07:
I don't recommend the "hi" functions, because they operate on the user interface. It is better to use the underlying "db" functions.

You can use `dbOpenCellViewByType` to open the cell view where you want to create the instance and the cell view you want to instantiate. Then use `dbCreateInst` to create the instance and use normal attribute access (`->` in skill or `.` in python) to read or change its parameters.

Please consult the cadence documents for details. It is expained much better there. I can only help you with skillbridge usage. But to me it seems you first need a working skill code.

You can come back when you have a specific problem with the skillbridge usage.

xiaohangguo, 2024-06-07:
I know that the interactive graphics window creation instance is not very good. But ,When creating an instance, whether it is `dbCreateInst` or `dbCreateParamInst`, the callback function will not be executed normally, for example, if I want to create a mom capacitor, its capacitance cannot be updated normally.

xiaohangguo, 2024-06-07:
[image]
I know you don't understand why you need to do this, but it is true that the two APIs `dbCreateInst` or `dbCreateParamInst` do not work very well, and I am also helpless.

xiaohangguo, 2024-06-07:
So I think the best solution is, can you give me a suggestion of a way to get through the skill bridge, close this window, now I need a mouse click to do it properly

TM90, 2024-06-07:
> I know that the interactive graphics window creation instance is not very good. But ,When creating an instance, whether it is `dbCreateInst` or `dbCreateParamInst`, the callback function will not be [...]

The best option here is to search the cadence support on how to trigger callbacks via skill.

There is a `CCSinvokeCdfCallbacks.il` cadence provides which can be found there. This method can then be triggered via skillbridge:

```
ws["CCSinvokeCdfCallbacks"](...)
```

xiaohangguo, 2024-06-07:
> The best option here is to search the cadence support on how to trigger callbacks via skill.
>
> There is a `CCSinvokeCdfCallbacks.il` cadence provides which can be found there. This method can then be [...]

i dont know how to use

TM90, 2024-06-07:
> i dont know how to use

What do you mean how to use the from cadence provided method or cadence support?

xiaohangguo, 2024-06-07:
> What do you mean how to use the from cadence provided method or cadence support?

`CCSinvokeCdfCallbacks` method can't run in my CIW

TM90, 2024-06-07:
> `CCSinvokeCdfCallbacks` method can't run in my CIW

It is not available by default `CCSinvokeCdfCallbacks.il` is a file which can be aquired via the [cadence support site](support.cadence.com/). Somebody at your company or university should have a support login.
In there CCSinvokeCdfCallbacks is defined and can be used after the file is loaded.

xiaohangguo, 2024-06-07:
Can you share this file? Our company is developing on a server, and I don't have an account as an employee

TM90, 2024-06-09:
> Can you share this file? Our company is developing on a server, and I don't have an account as an employee

I am not allowed to share the file since it is owned by Cadence

closed 2024-06-13 by TM90