unihd-cag/skillbridge #268: [SUPPORT] Failes to do check&save

kakbydima, 2024-11-05:
Updated parameters of a cell's schematic, but check and save via skillbridge does not seem to perform specifically check the netlist and ADE failes to generate netlist.
The basic skill code does the job (schCheck(cv) + dbSave(cv))

With skillbridge there is an indication of schematic save (same as with skillcode), but no indication of schematic check being performed. 
Do you know how to check that the ws.db.check(cv) works properly?

The python code part is the following:
cv = ws.db.open_cell_view_by_type("lib", "cell", "celltype",  "", "a")
...
ws['CCSinvokeCdfCallbacks'](cv, order=["l", "fingers", "fw", "w"])
ws.db.check(cv)
ws.db.save(cv)
ws.db.close(cv)

nielsbuwen, 2024-11-05:
The correct way to translate `schCheck(cv)` into Python would be `ws.sch.check(cv)`.

kakbydima, 2024-11-05:
Hi nielsbuwen, 
got it, thank you for the response!

nielsbuwen, 2024-11-05:
You are welcome. Did that solve your issue? If yes, you can close the issue

kakbydima, 2024-11-05:
Yes, it does solve the issue. Thank you

closed 2024-11-05 by kakbydima