unihd-cag/skillbridge #269: [SUPPORT] How to create a global data in python?

xiaohangguo, 2024-11-19:
I want to create a global variable in python and call it somewhere else, is this possible?
```
    def create_hash_summary_table(
        self,
    ):
        sum_hashtable = self.workspace["makeTable"]("sum_hashtable")
        cv = self.workspace.ge.get_edit_cell_view()
        incomplete_set = self.workspace.ge.create_hilight_set(cv,["y3" "drawing"])
        return
```

```
    def highlight_point_list(self,id:str, bbox :list):
        self.workspace.__.incomplete_set
```

nielsbuwen, 2024-11-19:
Hi, you closed the issue right away. Was that intended? Or do you still need assistance?

xiaohangguo, 2024-11-20:
> Hi, you closed the issue right away. Was that intended? Or do you still need assistance?

I'll use global variables now, but accessing and iterating is a bit problematic, but I'm avoiding it for now.

```
    def create_hash_summary_table(self):
        # 创建全局变量命名空间
        globals = self.workspace.globals("my_globals")
        Console.log(globals)
        # 创建哈希表并存储到全局变量中
        globals.sum_hashtable << self.workspace["makeTable"]("sum_hashtable")

        # 获取编辑中的 CellView
        cv = self.workspace.ge.get_edit_cell_view()
        hilight_set = self.workspace.ge.create_hilight_set(cv, ["y3", "drawing"])
        hilight_set.enable = True
        # 创建高亮集并存储到全局变量中
        globals.incomplete_set << hilight_set
        Console.log(globals)
```

```
    def highlight_point_list(self, id: str, bbox: list):
        "bbox :[x1,y1,x2,y2]"
        globals = self.workspace.globals("my_globals")
        x1 = bbox[0] / 1000.000
        y1 = bbox[1] / 1000.000
        x2 = bbox[2] / 1000.000
        y2 = bbox[3] / 1000.000
        rect_path = [[x1, y1], [x2, y1], [x2, y2], [x1, y2], [x1, y1]]
        # 访问全局变量 incomplete_set
        incomplete_set = globals.incomplete_set

        # globals.incomplete_set.for_each(path_db.append(skillbridge.loop_var))
        # # for item in incomplete_set:
        # #     path_db.append(item)
        # globals.sum_hashtable[id] = path_db
        # Console.log("highlight line db:", globals.sum_hashtable[id])
```