unihd-cag/skillbridge #271: [FEATURE] transfer a waveform family from skill to python

dinapolir, 2024-12-18:
Being able to transfer a waveform (or better a waveform family) from skill to python, for example as a numpy array, would be very interesting to make data analysis of the waveforms in python.

TM90, 2024-12-19:
Hi,

Extracting y and x data is possible via the following skill methods which can be accessed via skillbridge return value should be a python array which you can use to create your numpy array.

` ws.dr.get_waveform_y_vec(...)` and  `ws.dr.get_waveform_y_vec(...)`

See: https://github.com/unihd-cag/skillbridge/issues/261

A more performant way when exporting a lot of data is to use the skill methods to write waveforms to a .csv file.
Afterwards you can use pandas to read the csv and to extract rows or columns as numpy arrays

Look in the Virtuoso Visualization and Analysis XL SKILL Reference for `awvSaveToCSV` which would translate to `ws.awv.save_to_csv(...)`

TM90, 2024-12-30:
I hope this helps if there is a follow up or open questions feel free to reopen the issue

closed 2024-12-30 by TM90

dinapolir, 2024-12-30:
Hi,
It helps, thanks.
Actually, the best would be to be able to open directly cadence native simulation files in Python but I didn't find anything and I suppose it is not possible.
Thanks again and kind regards,
Roberto

Inviato da Outlook per Android<https://aka.ms/AAb9ysg>