unihd-cag/skillbridge #273: [BUG] pyStartServer errors when 'singleMode argument is used.

jasonpjacobs, 2025-02-24:
**Describe the bug**

When calling the SKILL function pyStartServer with singleMode enabled, the Python call errors:

     python_server.py: error: argument log_level: invalid choice: 'INFO--single' (choose from 'DEBUG', 'WARNING',     'INFO', 'ERROR', 'CRITICAL', 'FATAL')

I've fixed this locally by editing  "skillbridge/server/python_server.il" and adding a space to the "--single" argument (line 205 in the 1.7.2 version).

**To Reproduce**
Steps to reproduce the behavior:

    pyStartServer(? id "default" ?singleMode 't)

TM90, 2025-02-26:
It seems we have a bug for the  single mode option. when creating the argument for the python part a whitespace is missing. 

I will recreate the issue on my side and afterwards prepare a Pull request

nielsbuwen, 2025-02-26:
Here is a PR: #274

TM90, 2025-02-26:
PR is merged and released with skillbridge 1.7.3

To apply the update run:

```
pip install --upgrade skillbridge
```