unihd-cag/skillbridge #276: [BUG] newbie's question

gsohler, 2025-03-17:
**Describe the bug**
Cannot install skillbridge

**To Reproduce**
Steps to reproduce the behavior:

I tried like documented
git clone this repo
cd skillbridge
pip install -e .

**Expected behavior**
skippbridge gets installed

**Error Stacktrace**

Instead, there i see:

[gsohler@rome skillbridge]$ pip install -e .
Directory '.' is not installable. File 'setup.py' not found.
[gsohler@rome skillbridge]$ 

**Additional context**
Add any other context about the problem here.

nielsbuwen, 2025-03-17:
Also the error message you got from `pip install -e .` indicates that your version of pip is very outdated

gsohler, 2025-03-17:
thank you for your fast repsonse.
I have already translated 100 lines of my skill script to python. 
its much easier to read and just great!
thank you for your effort!

closed 2025-03-17 by gsohler

gsohler, 2025-03-18:
Hey and, I love how great the skill bridge is linked to python, even with properties, versatile function name mapping , arrays , and everything!
only thing which it cant to out of the box is this:
views =  self.ws.dd.get_obj(libname, cellname).views.name

but fortunately , python has list comprehension

But I think i found another bug/irregularity:

I am opening a cellview in the bridge which is already open with main instance of Virtuoso:

self.cv=self.ws.db.open_cell_view_by_type(libname, cellname, "schematic", "schematic", "w")   

it returns something usable and i do something with it.

 self.ws.sch.check(self.cv)
 print(self.cv.mode) # returns "w" 
 self.ws.db.save(self.cv)

# however, this save does not work, it returns an error:

   self.ws.db.save(self.cv)
  File "/home/gsohler/venv/lib64/python3.6/site-packages/skillbridge/client/functions.py", line 41, in __call__
    result = self._channel.send(command)
  File "/home/gsohler/venv/lib64/python3.6/site-packages/skillbridge/client/channel.py", line 157, in send
    return self._receive_only()
  File "/home/gsohler/venv/lib64/python3.6/site-packages/skillbridge/client/channel.py", line 153, in _receive_only
    return self.decode_response(response)
  File "/home/gsohler/venv/lib64/python3.6/site-packages/skillbridge/client/channel.py", line 47, in decode_response
    raise RuntimeError(response)
RuntimeError: ("putprop" 0 t nil ("*Error* putprop: first arg must be either symbol, list, defstruct or user type" nil))

IMHO open_cell_view_by_type should have returned None, as we cannot get exclusive write permissions to a cellview, which is already in use.

Is there anything which i am doing wrong ?