unihd-cag/skillbridge #279: [SUPPORT] How to run simulation using ocean command in the SKILLBRIDGE

rhuu16, 2025-04-21:
Can I know how to run ocean command in the SKILLBRIDGE? 
from skillbridge import Workspace
This how I converted Ocean Script to python but there is some issue with analysis I guess. Is that any correct way?
# Open SkillBridge session
ws = Workspace.open()

# Set simulator and design
ws['simulator']('spectre')
ws['design']("testsimu1", "idtest", "schematic")
ws['analysis']('dc', saveOppoint=True)
# Design variables
ws['desVar']("vdd", 1.8)
ws['desVar']("vgs", 1)
ws['envOption']('analysisOrder',list("dc"))
# # Save current through /M0/D
ws['save']('i', "/M0/D")
# Run simulation
ws['run']()

nielsbuwen, 2025-04-26:
Hi,

sorry for the late response. Can you please share the original SKILL code with us. Then i can take a look if something went wrong while translating it to python.

rhuu16, 2025-04-26:
Hello there!, thank you for the response. This above is the skill code :
```
design("library" "celview" "schematic")
analysis('dc ?saveOppoint t  ?param "vdd"  ?start "0"  
		?stop "2.0"  ?step "0.2"  )
desVar(	  "vdd" 1.8	)
desVar(	  "vgs" 1	)
envOption(
	'analysisOrder  list("dc") 
)
save( 'i "/M0/D" )
temp( 27 ) 
run()
selectResult( 'dc )
plot(getData("/M0/D") )
```

rhuu16, 2025-04-27:
Yeah its working thank you so much!