unihd-cag/skillbridge #280: [SUPPORT] Kindly guide me how to change this skillcode to python

rhuu16, 2025-07-28:
May I know how to translate this skillcode to python? , it will be very helpful thanks!

`\1\0\0\.\*\(p\(PORT0\:r\ \(\/net6\ \/gnd\!\)\)\-p\(\/PORT1\/PLUS\ \(\/net9\ \/gnd\!\)\)\)\/Pdc\(2\)\ \ hb\ P\/P = ((100.0 * harmonic((let(((vn (v("/net6" ?result "hb_fd") - 0.0))) spectralPower((vn / resultParam("PORT0:r" ?result "hb_fd")) vn)) + spectralPower(i("/PORT1/PLUS" ?result "hb_fd") (v("/net9" ?result "hb_fd") - 0.0))) '(1))) / (- harmonic((spectralPower(i("/V1/PLUS" ?result "hb_fd") (v("/net5" ?result "hb_fd") - 0.0)) + spectralPower(i("/V0/PLUS" ?result "hb_fd") (v("/net2" ?result "hb_fd") - 0.0))) '(0))))`

`Output\ Referred\ IP3\ Points = ipnVRI((v("/net6" ?result "hbac") - 0.0) '-2 '0 ?rport resultParam("PORT0:r" ?result "hbac") ?epoint 0 ?psweep nil ?measure "Output")`

rhuu16, 2025-07-28:
This how i converted into python: 
`hb_p_metric = (100.0 * ws['harmonic']((ws['v']("/net6", result="hb_fd") / ws['resultParam']("PORT0:r", result="hb_fd"), ws['v']("/net6", result="hb_fd")) +
    ws['spectralPower'](ws['i']("/PORT1/PLUS", result="hb_fd"), ws['v']("/net9", result="hb_fd"))
, [1])) / -ws['harmonic'](
    ws['spectralPower'](ws['i']("/V1/PLUS", result="hb_fd"), ws['v']("/net5", result="hb_fd")) +
    ws['spectralPower'](ws['i']("/V0/PLUS", result="hb_fd"), ws['v']("/net2", result="hb_fd"))
, [0])`
but I have error :
`Traceback (most recent call last):
  File "/home/ic/Downloads/newrl_asha (backup)/singleagent/fulltest.py", line 73, in <module>
    hb_p_metric = (100.0 * ws['harmonic']((ws['v']("/net6", result="hb_fd") / ws['resultParam']("PORT0:r", result="hb_fd"), ws['v']("/net6", result="hb_fd")) +
                                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
TypeError: unsupported operand type(s) for /: 'RemoteObject' and 'float'`

rhuu16, 2025-07-29:
This is the original expression I’m trying to convert:

```scheme
\1\0\0\.\*\(p\(PORT0\:r\ \(\/net6\ \/gnd\!\)\)\-p\(\/PORT1\/PLUS\ \(\/net9\ \/gnd\!\)\)\)\/Pdc\(2\)\ 
hb P/P = ((100.0 * harmonic((let(((vn (v("/net6" ?result "hb_fd") - 0.0))) 
spectralPower((vn / resultParam("PORT0:r" ?result "hb_fd")) vn)) + 
spectralPower(i("/PORT1/PLUS" ?result "hb_fd") (v("/net9" ?result "hb_fd") - 0.0))) '(1))) / 
(- harmonic((spectralPower(i("/V1/PLUS" ?result "hb_fd") (v("/net5" ?result "hb_fd") - 0.0)) + 
spectralPower(i("/V0/PLUS" ?result "hb_fd") (v("/net2" ?result "hb_fd") - 0.0))) '(0))))

rhuu16, 2025-07-29:
This is how I converted it into Python:

```python
hb_p_metric = (
    100.0 * ws['harmonic'](
        (
            ws['v']("/net6", result="hb_fd") / ws['resultParam']("PORT0:r", result="hb_fd"),
            ws['v']("/net6", result="hb_fd")
        ) + ws['spectralPower'](
            ws['i']("/PORT1/PLUS", result="hb_fd"),
            ws['v']("/net9", result="hb_fd")
        ),
        [1]
    )
) / -ws['harmonic'](
    ws['spectralPower'](ws['i']("/V1/PLUS", result="hb_fd"), ws['v']("/net5", result="hb_fd")) +
    ws['spectralPower'](ws['i']("/V0/PLUS", result="hb_fd"), ws['v']("/net2", result="hb_fd")),
    [0]
)

nielsbuwen, 2025-07-29:
Should we implement operator overloading in skillbridge? Then `ws['v'](...) / ws['resultParam'](...)` should just work.

rhuu16, 2025-07-30:
### 📌 Problem: `wave2vec` returns RemoteObject list, not actual float values

Hi there — thanks for the help so far! I’m running into an issue with `wave2vec`.

Here’s what I currently have:

```python
vout = ws['v']("/net6", result="hb_fd")
rload = ws['resultParam']("PORT0:r", result="hb_fd")
print(vout)

# Convert to vectors
_, vout_vals = wave2vec(ws, vout)
print(vout_vals)

And here’s the output I get when running the script:
<remote srrWave@0x3ef830e0>
[<remote object@0x3ef83600>, <remote object@0x3ef83620>, …, <remote object@0x3ef83870>]

TM90, 2025-07-30:
It is always difficult to debug/unserstand these issues since we cannot reproduce this case on our side as we do not have the same data/ simulation setup.

I can offer to create a minimal example on how to extract waveform data on my side and share the code. This will take some time but should happen during this week.

TM90, 2025-08-03:
I did not get to it this week but maybe this can issue can point you towards the solution?

https://github.com/unihd-cag/skillbridge/issues/271

rhuu16, 2025-08-04:
And the output I get is:
So vout is a srrWave, and when I pass it to wave2vec, the returned vout_vals is a list of RemoteObjects — not float values. I assume I still need to explicitly dereference or cast them with ws.dr.get_elem(...)?

Let me know if that’s expected or if there’s a preferred way to get concrete float vectors from srrWave signals.

Thanks again!

```python
<remote srrWave@0x3ef830e0>
[<remote object@0x3ef83600>, <remote object@0x3ef83620>, ..., <remote object@0x3ef83870>]

TM90, 2025-08-14:
> And the output I get is: So vout is a srrWave, and when I pass it to wave2vec, the returned vout_vals is a list of RemoteObjects — not float values. I assume I still need to explicitly dereference or cast them with ws.dr.get_elem(...)?
> 
> Let me know if that’s expected or if there’s a preferred way to get concrete float vectors from srrWave signals.
> 
> Thanks again!
> 
> <remote srrWave@0x3ef830e0>
> [<remote object@0x3ef83600>, <remote object@0x3ef83620>, ..., <remote object@0x3ef83870>]

I guess this can be expected. The output highly depents on the structure of your `srrWave` object. 

That is also the reason why this is difficult to reproduce.